OpenDingux/buildroot #22: opendingux: rg350m: broken apps/games

jezze, 2020-10-02:
Most of these problems seem to be rg350m specific.

APPS:

wallpaper: does not start
inputtester: turns the screen black and dies.

GAMES:

apricots: does not start
griffon legend: does not start
powder: visual artifacts
puzzletube: does not start
samego: does not start
snowman: does not start
spout: cant be exited?
stringrolled: does not start
umg: displays a cursor during boot

EhotPotaskun, 2020-12-25:
ST Terminal and FCEUX not starting too.

I've got strace logs of each application, where working one is from HDMI Beta firmware where it works, and another one from firmware that was built with latest buildroot from git an it fails to start.

_(A lot of KDGKBENT events at the end of the log trimmed for smaller size)_

[st_working.log](https://github.com/OpenDingux/buildroot/files/5742297/st_working.log)
[st.log](https://github.com/OpenDingux/buildroot/files/5742294/st.log)
[fceux_working.log](https://github.com/OpenDingux/buildroot/files/5742295/fceux_working.log)
[fceux.log](https://github.com/OpenDingux/buildroot/files/5742296/fceux.log)

pcercuei, 2020-12-30:
Some games (puzzletube, snowman, samego) are just a packaging problem and the OPKs can be fixed without recompiling anything.
Powder was fixed.
Stringrolled is Python2 and needs to be ported to Python3 (alternatively, I have an experimental OPK that packs Python2 inside).
About SDL2, the problem is known and is high on the list.

JORGETECH, 2022-03-16:
@jezze I made a list on OpenDingux Beta compatibility for game ports. I found that around 49 of the titles I tested (168 in total) are broken in some way, since a bunch of them fail due to missing libraries I'll try to repackage them with libraries from older OpenDingux buildroot or other suitable replacement. You can check my list here: https://github.com/JORGETECH/od-beta-ports-compat-list-spreadsheet

glebm, 2022-11-21:
> ST Terminal and FCEUX not starting too.

Recompiled ST Terminal is available here https://github.com/od-contrib/st-sdl/releases

Note that the LD_LIBRARY_PATH fix is to specify it to the current working directory explicitly, i.e.:

```bash
LD_LIBRARY_PATH="$(pwd)" ./st
```

a-dekker, 2022-12-21:
Almost all games/apps should be adapted for Opendingux-Beta by now, see also https://github.com/eduardofilo/RG350_adam_image/wiki/En:-8.-Compatible-game-ports
So I would consider this issue "solved" meanwhile.