OpenDingux/buildroot #23: opendingux: sdl2: apps using sdl2 does not work

jezze, 2020-10-03:
No information yet on why this is.

jezze, 2020-10-03:
Here is the little working testprogram I did. Build instructions found in readme.

https://github.com/jezze/testsdl2

jezze, 2020-10-04:
I have now checked what the drivers are:

Driver 0: opengles2
Driver 1: opengles
Driver 2: software

So this tells me opengles works, and opengles2 does not.

jezze, 2020-10-08:
There are also framedrops going on. Seems to be a related to vsync. Adding SDL_GL_SetSwapInterval(1) fixes that problem.

KatyushaScarlet, 2020-10-21:
Same issue with rg350m, using 640x480 resolution.
The screen just turn black and the device is stuck.

pcercuei, 2021-03-22:
It should work out-of-the-box now; please test. Use `-1` as the render driver so that it's picked automatically.

jezze, 2021-03-24:
Yes this works now. Tried on my RG350M. Nice work!

closed 2021-03-24 by jezze