OpenDingux/buildroot #25: opendingux: rg350(m): gmenu2x can't read charger status

jezze, 2020-10-09:
The path to the power supply is different from how it was on the GCW Zero.

acarteron, 2020-10-18:
gmenu2x can handle battery status using voltage_max_design  voltage_min_design  voltage_now files.
The issue is that voltage_now might contains a wrong value that never increase or decrease with battery charge.
But when an application that not use joystick (e.g. Terminal) is used, the value increase or decrease with battery charge even if it does not correspond to real battery voltage.

In gmenu2x:
```
rg350m:~ $ cat /sys/class/power_supply/jz-battery/voltage_now 
2694000
```
Terminal application started:
```
rg350m:~ $ cat /sys/class/power_supply/jz-battery/voltage_now 
5170000
```

pcercuei, 2020-11-22:
I think it's fixed now in the latest nightly, correct?

closed 2021-01-19 by pcercuei