OpenDingux/buildroot #27: buildroot can't have python2 and python3 at the same time

glebm, 2020-10-14:
Some OPKs, such as [Skifree](https://github.com/retrogamehandheld/OpenDingux/raw/master/Games%20%26%20Ports/Skifree.opk), require python2.

Not sure if we want to support them or would rather have python3.

jezze, 2020-10-15:
My gut feeling is that we will not support python2. It's just too old and as you say buildroot doesn't even support having both at the same time.

michaelshmitty, 2020-11-19:
I also think no longer supporting Python 2 is a good idea. I'd be happy to look into porting anything that still relies on Python 2 to Python 3. From a quick glance at the source code of Skifree I see the only reason it's using Python 2 is because of the age of the code, it relying on urllib2 (dropped in Python 3 in favor of the requests module) and Pygame 1.9. But even Pygame's migration to version 2.0 now recommends Python 3. I'm sure I'd be able to port it to Python 3. If there are any other opk's or general OD need for Python 2, feel free to mention them here. Happy to look into it.
Thanks for your great work on OD! You can also find me on the RGH discord.

pcercuei, 2020-11-22:
I agree, Python3 was out a decade ago, it's not worth the effort to support both.

closed 2020-11-22 by pcercuei