OpenDingux/buildroot #56: "Updated rootfs is corrupted!"

dataforestLTD, 2021-05-18:
Every time I try to run the updated I get an "Updated rootfs is corrupted! Please report this bug"
It also looks like it says "Segmentation error" in the terminal before the error screen pops up.
Getting this on an RG-350M, have tried numerous times on freshly flashed firmware, and always the same result, have tried different SD cards, different versions of the updater, flashing new stock firmware with and without extending the ext4 partition, always the same error.

askotx, 2021-05-18:
I can confirm this issue too

pcercuei, 2021-05-31:
The installer copies the files then verifies that they were copied correctly, if you get this error, that probably means that the native firmware is corrupting the files.

If you run Linux, you can try flashing OD nightly over USB from the terminal using the odboot-client tool: https://app.box.com/s/qxz0n633wibai6yxbxqy3mn4ay7j2d3l

closed 2021-09-10 by pcercuei