OpenDingux/buildroot #62: [feature request] sleep mode LED on

Icing-weaponry, 2021-06-19:
Hi there,

I hope this is the proper place to ask for this but I have an RG280v and whevener it goes to sleep I forget about it so the battery dies or when I think it's in sleep I turn it on by accident, which is not too troublesome but still.

When I look at other systems like the 3ds or PS4 they keep their LED on or flashing when they enter sleep mode but when my RG280v there's no screen nor led, leading to the state confusions mentioned above.

I thought it would have been a good idea to have the led on while sleep that way it give a clear indication that if the screen if off but the LED is on, the device is currently sleeping.

thank you for your time

pcercuei, 2021-09-10:
Should be the case now.

closed 2021-09-10 by pcercuei

Icing-weaponry, 2021-09-11:
amazing, thank you so much!