OpenDingux/buildroot #72: Latest official toolchain is not portable

lephilousophe, 2021-11-12:
The latest toolchain downloaded from http://od.abstraction.se/opendingux/toolchain/ has been built on an unknown recent OS and the libraries it uses are too modern for a Debian Bullseye (latest stable) host.
Two possibilities here: either tell which platform can be used to run the toolchain or build the toolchain with an old enough platform to not having to take care of this.

In addition several libraries, not libc/gcc related, are not bundled in the lib subdirectory and require to be installed on the system:
```
libbz2.so.1.0
libgdbm_compat.so.4
libgdbm.so.6
libncursesw.so.6
libreadline.so.8
libunistring.so.2
libxxhash.so.0
```

It would be great to have a fully portable toolchain, that is with all needed libraries included and which could run on any system not too old but not too recent too.

pcercuei, 2021-11-19:
That would be something that the upstream Buildroot project would need to support.

If you prefer, there's also the option of shipping the toolchain inside a Docker image.

lephilousophe, 2021-11-20:
it could be useful yes as I already use the toolchain in a docker container.
Currently I build it from source.

pcercuei, 2021-11-21:
We use docker for building the toolchains in our CI: https://github.com/OpenDingux/RetroHandhelds-Docker

lephilousophe, 2021-11-21:
My main problem with CI builds is that it seems I cannot download them using wget while the ones on od.abstraction.se can be.
But I was under the impression that od.abstraction.se toolchains were built using an Arch or Fedora host and not a Ubuntu 20.04.
Do you confirm that od.abstraction.se toolchains are built with Ubuntu 20.04?

pcercuei, 2021-11-21:
I don't know, I don't manage od.abstraction.se.

lephilousophe, 2021-11-21:
Hum... I took the link from https://opendingux.net/ which also has a link to your github page. I assumed it was operated by you. Sorry!
My issue was related to these builds so I assume we can close this.

When you will be ready for release, if you could upload the toolchains on GiHub releases page once you tag, it would be great.

closed 2021-11-21 by lephilousophe