OpenDingux/buildroot #90: Move buildroot to a submodule

glebm, 2022-11-05:
OpenDingux buildroot is currently a fork of upstream buildroot, making merges/rebases a bit of work.
It'd be easier to maintain and upgrade if  buildroot was in a submodule pointing to upstream.

Some examples of doing it this way:

1. Batocera: https://github.com/batocera-linux/batocera.linux
2. RetroLX: https://github.com/RetroLX/retrolx

pcercuei, 2022-11-11:
I don't see the point. I have zero problems merging upstream Buildroot changes with the current system.

glebm, 2022-11-11:
In terms of organization, it's nice to have all our custom code cleanly separated, but it's a minor thing

closed 2022-11-11 by glebm

pcercuei, 2022-11-11:
But it is already cleanly separated - it is in `boards/opendingux/`

glebm, 2022-11-11:
Almost, `configs/od_*` and `(re)build.sh` are not in `boards/opendingux`