OpenDingux/buildroot #105: Playing mod music files often results in a crash

a-dekker, 2022-12-17:
Playing mod music files is largely broken on OD-beta. And it got even worse in the latest releases of OD-beta, some games with mod music that did work a few releases earlier now also crash. some people resort to including their self compiled library files (https://github.com/plrguez/hurrican-od/blob/main/Hurrican/build_libmodplug.sh for example) if possible, but that's an unwanted workaround if you ask me.
Any chance this could be addressed?

mthuurne, 2022-12-17:
Which library is used by SDL_mixer for playing MOD files by default? I know we had issues with libmikmod in the past and I excluded that from the build to force libmodplug to be used instead. Maybe those build modifications got lost over time.

plrguez, 2022-12-17:
If I'm not interpreting this wrong, running `readelf -d libSDL_mixer.so` on the library in the toolchain indicates that it uses `libmodplug.so.1`.

```
Dynamic section at offset 0x148 contains 30 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libSDL-1.2.so.0]
 0x00000001 (NEEDED)                     Shared library: [libiconv.so.2]
 0x00000001 (NEEDED)                     Shared library: [libdrm.so.2]
 0x00000001 (NEEDED)                     Shared library: [libudev.so.1]
 0x00000001 (NEEDED)                     Shared library: [libts.so.0]
 0x00000001 (NEEDED)                     Shared library: [libmodplug.so.1]
 0x00000001 (NEEDED)                     Shared library: [libstdc++.so.6]
 0x00000001 (NEEDED)                     Shared library: [libmad.so.0]
 0x00000001 (NEEDED)                     Shared library: [libc.so.0]
 0x0000000e (SONAME)                     Library soname: [libSDL_mixer-1.2.so.0]
 0x0000001d (RUNPATH)                    Library runpath: []
 0x0000000c (INIT)                       0x1d38
 0x0000000d (FINI)                       0x1d760
 0x00000004 (HASH)                       0x260
 0x00000005 (STRTAB)                     0x11a8
 0x00000006 (SYMTAB)                     0x718
 0x0000000a (STRSZ)                      2663 (bytes)
 0x0000000b (SYMENT)                     16 (bytes)
 0x00000003 (PLTGOT)                     0x279b0
 0x00000011 (REL)                        0x1c10
 0x00000012 (RELSZ)                      296 (bytes)
 0x00000013 (RELENT)                     8 (bytes)
 0x70000001 (MIPS_RLD_VERSION)           1
 0x70000005 (MIPS_FLAGS)                 NOTPOT
 0x70000006 (MIPS_BASE_ADDRESS)          0x0
 0x7000000a (MIPS_LOCAL_GOTNO)           175
 0x70000011 (MIPS_SYMTABNO)              169
 0x70000012 (MIPS_UNREFEXTNO)            32
 0x70000013 (MIPS_GOTSYM)                0x39
 0x00000000 (NULL)                       0x0
```

mthuurne, 2022-12-18:
Paul has identified [an issue with unaligned reads](https://github.com/OpenDingux/buildroot/pull/81#issuecomment-1356800585) that might explain libmodplug crashes.

pcercuei, 2022-12-21:
Should be fixed in this build:
https://github.com/OpenDingux/buildroot/actions/runs/3735504867

a-dekker, 2022-12-21:
Looks promising, games that recently stopped working like UMG and Dodgin' Diamond II can be played again!

a-dekker, 2022-12-21:
Previous issues with mods are still there (as I was a bit afraid of), but is addressed I just read https://github.com/eduardofilo/RG350_adam_image/discussions/191#discussioncomment-4470672

PR #122 (2023-05-06): [od] add libmodplug patches
These are patches for libmodplug that fix various UB issues. They come from these PRs to the upstream repository:
1. https://github.com/Konstanty/libmodplug/pull/92
2. https://github.com/Konstanty/libmodplug/pull/93

The upstream maintainer has not merged any PRs in over a year.

Fixes #105.
commit 80a6c809 (2023-05-06): [od] add libmodplug patches

These are patches for libmodplug that fix various UB issues.
They come from these PRs to the upstream repository:
1. https://github.com/Konstanty/libmodplug/pull/92
2. https://github.com/Konstanty/libmodplug/pull/93

The upstream maintainer has not merged any PRs in over a year.

Fixes #105.
closed 2023-05-06 by pcercuei