OpenDingux/buildroot #118: GCW0 - HDMI Out taking up some cpu cycles?

BrettLeeARV, 2023-02-18:
In HDMI out mode, audio intermittently stutters making me suspect that in this mode, some CPU cycles are being consumed. Even the FPS shows about 1-2 fps dip.

pcercuei, 2023-02-24:
@BrettLeeARV well it depends what app you are testing with.

If the app still renders at 320x240 it shouldn't eat more CPU cycles since it's scaled up in hardware. However, if it tries to make use of the 720p resolution, then it will spend much more time during rendering, yes.

BrettLeeARV, 2023-02-25:
@pcercuei Thanks for your reply. So far I'm using integer or hardware scaling options found in most emulators. In Retroarch, I'm using integer scaling option, which should make use of the hardware scaler. Pressing Power+B option toggles between the various hardware scaling modes.