OpenDingux/buildroot #119: Ben Nanonote support

wryun, 2023-03-02:
I pulled my Ben Nanonote out and was wondering if I could easily run anything other than the latest 'official' OpenWRT firmware from 2012.

Do you have a sense of how far OpenDingux is from working on a Nanonote, or any pointers on how to make it happen?

pcercuei, 2023-04-19:
You'd need a complete new "flavour" since the current ones ("rs90", "gcw0" and "lepus") don't support the JZ4740.

You'd also have absolutely no software available for it. So it wouldn't be very difficult but it'd be pretty much pointless.

wryun, 2023-04-20:
Thanks. Kernel still works with JZ4740 as as far as you know?

pcercuei, 2023-04-20:
The Ben Nanonote *should* still work with the upstream kernel, yes. But nobody tested in years, and therefore it is not a guarantee.