OpenDingux/buildroot #141: can't complete installation with latest build

Rumtrinker, 2023-09-09:
tried both installations on my retromini with the latest build ( 18-6-2023) . has an error. will try a earlier one i think. 
[image]
[image]

Rumtrinker, 2023-09-09:
retried like 4 times. got installed by some miracle. a lot of thanks.

pcercuei, 2023-09-14:
Looks like a USB issue, I'd have advised to try with a different cable and/or on a different computer. Good that you got it to flash eventually.

closed 2023-09-14 by pcercuei

Rumtrinker, 2023-09-14:
got it working after returning everything to the previous driver. also I
only got working RetroArch ( the only emulador needed) and Commander. other
apps doesnt work on the newest build. Thank you for your hard work.