OpenDingux/buildroot #145: RS90 transfer files isues

SamuelFL, 2023-12-13:
Are there known issues/fixes for windows data transfers?

pcercuei, 2024-04-03:
I don't know, you tell me. What's the issue?

ValdikSS, 2025-01-28:
It's not only Windows, it's also the issue on Linux.

MTP hangs in the end of large file transfer, requiring replugging the cable.

And it seems to be the issue with the USB stack/USB controller initialization, not with the MTP gadget.

As posted on IRC:

```
[09:13] <valdikss> So the USB is so broken in the latest kernel/bootloader that I'm unable to enter recovery mode with SELECT, have to disassemble the device
[10:40] <valdikss> It seems that the USB controller state is maintained accross CPU resets, the reset button won't reset the controller, and the bootrom also doesn't reset the controller in a proper way. Only booting proper kernel (which correctly initializes the controller) works, or unplugging the battery.
[10:40] <valdikss> Reproduction is simple: 1. Flash Adam image to the internal MicroSD; 2. Boot it; 3. Power off the device; 4. Unplug the internal MicroSD; 5. Try to connect the device in USB boot mode over USB
[10:40] <valdikss> The result:
[10:41] <valdikss>   +0,378797] usb 3-1: new full-speed USB device number 102 using xhci_hcd
[10:41] <valdikss> [  +0,000243] usb 3-1: Device not responding to setup address.
[10:41] <valdikss> [  +0,205022] usb 3-1: Device not responding to setup address.
[10:41] <valdikss> [  +0,207712] usb 3-1: device not accepting address 102, error -71
[10:41] <valdikss> [  +0,000120] usb 3-1: WARN: invalid context state for evaluate context command.
[10:41] <valdikss> [  +0,111930] usb 3-1: new full-speed USB device number 103 using xhci_hcd
[10:41] <valdikss> [  +0,000235] usb 3-1: Device not responding to setup address.
[10:45] <valdikss> 1. The kernel should disable power to USB controller upon shutdown
[10:46] <valdikss> 2. ubiboot should probably reset the controller to prevent this kind of issues as I have right now
```