OpenDingux/buildroot #148: RetroArch snes9x: `lxhu    v0,s0,v0,0` SIGILL, Illegal instruction

ValdikSS, 2025-01-30:
RetroArch [v1.20.0](https://buildbot.libretro.com/stable/1.20.0/dingux/mips32-odbeta/RetroArch.7z) (or [2025-01-30_RetroArch.7z](https://buildbot.libretro.com/nightly/dingux/mips32-odbeta/2025-01-30_RetroArch.7z) nightly) `mips32-odbeta` snes9x core is crashing in ingenic `lxhu` instruction.

They use [opendingux-gcw0-toolchain.2022-09-22.tar.xz](https://git.libretro.com/libretro-infrastructure/libretro-build-dingux/-/tree/master/toolchain) to build odbeta build, which includes commit b7b803bdcb718ea6f63a268a7462dd011d1c977d.
Non-odbeta (non-ingenic instructions enabled toolchain) works fine.

### Files
[snes9x2005_libretro.zip](https://github.com/user-attachments/files/18601010/snes9x2005_libretro.zip) from the nightly build + [core.709.zip](https://github.com/user-attachments/files/18601179/core.709.zip) **corefile**.

### Text

```
gcw0:~ $ gdb --args /usr/local/bin/retroarch_rg350_odbeta -L /media/data/local/home/.retroarch/cores/snes9x2005_libretro.so /media/data/roms/SFC/Aladdin\ \(USA\).sfc
GNU gdb (GDB) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "mipsel-gcw0-linux-uclibc".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/local/bin/retroarch_rg350_odbeta...
(No debugging symbols found in /usr/local/bin/retroarch_rg350_odbeta)
(gdb) r
Starting program: /media/data/local/bin/retroarch_rg350_odbeta -L /media/data/local/home/.retroarch/cores/snes9x2005_libretro.so /media/data/roms/SFC/Aladdin\ \(USA\).sfc
[INFO] === Build =======================================
[INFO] Capabilities: Т�w�F�wТ�w�K� �w���w�K�ȁ�w���w��������`t�III�
[INFO] Version: 1.20.0
[INFO] Built: Jan 30 2025
[INFO] =================================================
[New LWP 712]
[INFO] [Input]: Found input driver: "sdl_dingux".
[INFO] [Core]: Loading dynamic libretro core from: "/media/data/local/home/.retroarch/cores/snes9x2005_libretro.so"
[INFO] [Overrides]: Core-specific overrides found at "/media/data/local/home/.retroarch/config/Snes9x 2005/Snes9x 2005.cfg".
[INFO] [Config]: Loading config: "/media/data/local/home/.retroarch/retroarch.cfg".
[INFO] [Config]: Appending override config: "/media/data/local/home/.retroarch/config/Snes9x 2005/Snes9x 2005.cfg".
[INFO] [Overrides]: Redirecting save file to "/media/data/local/home/.retroarch/saves/Snes9x 2005/Aladdin (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/media/data/local/home/.retroarch/states/Snes9x 2005/Aladdin (USA).state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2_INTL.
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[libretro INFO] Frontend supports RGB565 - will use that instead of XRGB1555.
[INFO] [Content]: Loading content file: "/media/data/roms/SFC/Aladdin (USA).sfc".
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_AUDIO_BUFFER_STATUS_CALLBACK.
[INFO] [SRAM]: Skipping SRAM load.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 256x224, Aspect: 1.333, FPS: 59.92, Sample rate: 32040.00 Hz.
[INFO] [Video]: Set video size to: 320x240.
[INFO] [Audio]: Set audio input rate to: 32081.31 Hz.
[New LWP 713]
[INFO] [Joypad]: Found joypad driver: "sdl_dingux".
[INFO] [Video]: Found display server: "null".
[INFO] [Audio]: Set audio input rate to: 32081.31 Hz.
[INFO] [ALSA]: Using ALSA version 1.2.5.1
[INFO] [ALSA]: Using FLOAT_LE sample format for PLAYBACK device "default"
[INFO] [ALSA]: Period: 4 periods per buffer (768 frames, 6144 bytes)
[INFO] [ALSA]: Buffer size: 3072 frames (24576 bytes)
[INFO] [ALSA]: Can pause: no.
[INFO] [ALSA]: Initialized PLAYBACK device "default"
[INFO] [Audio]: Started synchronous audio driver.
[... 119 lines cut ...]
            t0       t1       t2       t3       t4       t5       t6       t7
 R8   0000028f 00000000 00000005 00000001 00000003 00000003 00000000 776dc274 
            s0       s1       s2       s3       s4       s5       s6       s7
 R16  00000073 776dc2c4 776dc194 7763c070 77638590 77669ba4 ffffff80 776dbbe4 
            t8       t9       k0       k1       gp       sp       s8       ra
 R24  00000014 775d8c04 00000001 00000000 77644460 7feb49b8 776dc158 775ac230 
        status       lo       hi badvaddr    cause       pc
      00000413 00003210 00000000 006b33db 00800010 775d8ca8 
          fcsr      fir  restart
      13800024 00330000 00000000

```

Disassembling with toolchain's `objdump`:

```
   46c9c:       14600024        bnez    v1,46d30 <retro_set_environment-0x2fc38>
   46ca0:       00000000        nop
   46ca4:       3210ffff        andi    s0,s0,0xffff
   46ca8:       72021168        lxhu    v0,s0,v0,0   ← ← ← ← ← ← ← ← ← ←
   46cac:       8fbf001c        lw      ra,28(sp)
   46cb0:       8fb00018        lw      s0,24(sp)
   46cb4:       03e00008        jr      ra
   46cb8:       27bd0020        addiu   sp,sp,32
```

ValdikSS, 2025-01-30:
This is on JZ4770, GCW Zero.

ValdikSS, 2025-01-30:
@SiarheiVolkau, please take a look.

SiarheiVolkau, 2025-01-30:
Looks like unaligned memory access.
@ValdikSS first of all, please make sure your kernel has the following patch applied
[MIPS: uaccess: emulate Ingenic LXW/LXH/LXHU uaccess](https://lore.kernel.org/r/linux-mips/20230604122655.69698-1-paul@crapouillou.net/T/#t)

ValdikSS, 2025-01-30:
@SiarheiVolkau yes, indeed.
OpenDingux Beta 2022-09-22 includes gcc patch, but not unaligned access kernel patch.

SiarheiVolkau, 2025-01-30:
However, even if the issue is in the kernel patch, it's not good for performance to leave RA as is. It will be good to make a fix for it as well.